DerSpatz/PCB-reflow-solder-heat-plate #5: Idea: use an NTC thermistor for sensing the heat-plate temperature

DominikPalo, 2022-11-09:
The current design uses an LMT85 analog temperature sensor as the main sensor. As this sensor is not directly attached to the heated area (and can't be due to its limited temperature range –50°C to 150°C), the temperature measurements are not very precise and tend to be skewed/affected by the airflow around the sensor.

So, my idea is to use a 100K NTC thermistor instead. The thermistor would be attached directly to the bottom side of the heating area (by a piece of Kapton tape). Such NTC thermistors are very popular in 3D printers for sensing both the hot end and heated bed temperatures and therefore cheap and widely available. They are precise and can easily measure temperatures up to 300°C, so I think they would be a right fit for this project too. 

The connection to the MCU is very simple - just a couple of resistors and a capacitor connected to a single ADC input. Here is an example connection from the [BTT SKR Mini E3 board](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V1.2/BTT%20SKR%20mini%20E3%20V1.2sch.pdf):
[image]

gifter77, 2023-03-19:
It's pretty easy to do without any change to the PCB. Just connect your NTC thermistor between temp3 data and GND pins, optionally change R10 to something that works better as a resistance divider with your thermistor and change the temperature calculation code.
See how I have done it below. I'm using a 10K 3950 thermistor and I've changed R10 to 1k instead of 4.7K to get a better range in the ADC.
[image]

jgoldwein, 2023-03-20:
Great idea.  Have you changed the code too?  If so, what changes did you make?

VTeselkin, 2023-05-18:
Great idea, has anyone made changes to the code?