DerSpatz/PCB-reflow-solder-heat-plate #8: Low resistance Heatplates - not getting hot

ManCloud, 2023-01-19:
Hello, 

I have a little Problem: i ordered 15 boards qt JLCPCB and they all seem to have too low resistance values. 
I measured boards with about 1.34 ohms which work kinda fine but also have boards at 1.12ohms that do not.

If i put the resistance inside the firmware to 1.10ohm, the board will heat up until about 70°C and then constantly reduce the current until about 250mA and that's it...

and it i put the resistance to 1.12 ohm the board will surpass the 5A mark, which somehow makes my PSU shutting down (even though the >1.34ohm boards also surpass 5A but without shutting down).

I'm kinda clueless where to start debugging here 😅 

This was part of a "community order" and not a "for profit thing".

bwack, 2023-02-24:
How is it going with the project. I'm about to build this myself. No expereince, though did you measure using a four wite technique? (Two wires Pass a constant current, two wires measure voltage)

ManCloud, 2023-02-24:
No, I modified a resistance meter I used in the past for coil resistance measurements for my vape

bwack, 2023-02-24:
I don't see any replies from the maintainer here. Maybe I can have a look at the code this weekend.

ManCloud, 2023-02-24:
That would be nice! I already wanted to have a look myself but am pretty busy with other stuff 🙈

bwack, 2023-03-03:
I'm experiencing overcurrent. I'm feeding in 12V from my lab power supply set to limit current at 5A. It doesn't take long for the regulator to go beyond that limit.
I've had no luck in getting the serial debugging data, so I display the PWM value on the screen in real time. It shows that the PWM value drops down to 0 (100% power)  after just a few maybe 10 seconds.. 12V into ~1 Ohm is way too much.
Not sure what to do about this yet. The board does monitor the input voltage. I could try in software to find out what how low PWM can get before the supply voltage starts to drop, with consideration that cable and contacts will drop a lot of voltage as well.

bwack, 2023-03-07:
My video on this board. https://youtu.be/bX83ZVMr6Pg

Findings. Resistance is too low to produce heat up quick enough at sensible current levels (5A). This cannot be fixed without reducing the trace width, or putting more current into and removing the 5A PTC fuse. There is no current sensing.

 It goes way beyond 5A. My bench powersupply droppet the voltage to 7V at 5A constant current  though rised to 10V when melting tin hot. But slowly. The 7V drops to about 6V at the heating bed.

I was thinking of doing my own board by tuning the heatbed to accept 24 or perhaps 19V (60W lsptop PSUs) to get the current down to 3 or 2.5A. Adding current sense. Measure resistance of the bed in real time to calculate temperature. But people are too negative, and so limited use case, I don't bother with this project I think.