DerSpatz/PCB-reflow-solder-heat-plate #9: Tempted to start: is someone working on the issues on the to-do-list? Just to know.. Many thanks!

ik0adr, 2023-02-24:
Many thanks!!

bwack, 2023-02-24:
I'm finally building the hotplate 1.1 after a long pause. I'll give some feedback If I find anything worth mentioning. Ps I'm planning to do a youtube video as well.

ik0adr, 2023-03-01:
I am just waiting for PCB and components. 
Will appreciate youtube video!!
Cheers

ik0adr, 2023-04-03:
Hi all.
I am waiting for some components to finish the DerSpatz plate. 
I do have some questions thow..
Is the firmware on GitHub doing the reflow curve control? 
In readme they say the temperature sensing is incorrect.. 
I thought the thermistor mod would solve that. 
Am I wrong? 
Cheers, have a nice day

Berkays, 2023-05-21:
Feel free to check out my stm32 board implementation at [repository](https://github.com/Berkays/ReflowPlate) which should hopefully has a more organized source structure and some improvements like ntc, rotary encoder input etc.