DerSpatz/PCB-reflow-solder-heat-plate #11: Made slight code changes to take advantage of Red LED... plus some design suggestions

jgoldwein, 2023-03-19:
Hello and thank you for this amazing project.  I have now built 5 of these, all using JLCPCB boards and LSCS components that I have been able to solder on to the boards.  I have also made some slight improvements in the firmware code which I have forked and posted at https://github.com/jgoldwein/PCB-reflow-solder-heat-plate/blob/main/Firmware/pcb_reflow_fw/pcb_reflow_fw.ino.

A few suggestions:
1. Primarily, if there were a way to **separate/detach the heating surface from the primary electronics** (not including the temperature sensor), the utility of the reflow plate would increase significantly.  I find that after about 5-10 uses the heating panel deteriorates beyond its usefulness.   And, since the Dallas 1-wire temperature sensor is not being used, replacement would only require LMT85 replacement provided that the LED and R6,7 could be moved on to the main circuit board.
2. I did see a comment suggesting substitution of parts such as 2 separate LEDs in place of the one bi-LED component.  Makes sense to me, but probably relatively easy to do without changing the PCB layout.
3. No problem from my perspective using the ATMEGA4809-AFR.  Readily available and inexpensive from many vendors such as DigiKey and Mouser in my area (USA).  Took a bit of work to figure out the UPDI programming, but a positive experience for me.

Thanks again for designing this.  It is really appreciated and has been extremely useful to me.

JG

gifter77, 2023-03-19:
I believe the heating panel deteriorates quickly because it actually gets way hotter than what the LMT85 measures. Even when set to 140C I measured over 200C on the heatplate. An easy solution is to use an NTC thermistor stuck under the heatplate instead of the LMT85.

jgoldwein, 2023-03-19:
Good idea, but I know that solder melting for reflow does not seem to occur until after 1:40 or so for solder that has a 138° melting point on my system. I think we need to factor in the r values of any material such as PCBs that are in involved. 

Still, I think it would be a good idea if we could detach the heating plate and replace it easily rather than the current design. With constant heating, cooling, and reheating, I can't help but think that there will inevitably be some deterioration of the heating plate.

Thank you kindly for responding to my comment. I think again that this is a great product and just want to make it even better.