DerSpatz/PCB-reflow-solder-heat-plate #12: How many heating cycles does the PCB survive?

manuel-rw, 2023-05-06:
Hello everyone,
I want to get into SMD soldering.
Can anyone tell me, how many cycles / thermal stress this PCB usually survives, until it deteriorates?
I know that many factors will influence this, but it would be useful to get a general idea what to expect.

Thank you!
Best regards

manuel-rw, 2023-05-09:
Hi, thanks for the reply.
Since the heating PCB is not replaceable, what endurance are you getting with the IR?
It shocked me seeing people mentioning only 5 cycles, but I'm not sure how accurate that is.

bwack, 2023-05-09:
I think the 5 cycles is due to the huge temperature swings causing it to overheat. 

I haven't used mine more than maybe 10 times, but the pcb is still holding up. no warping.

bwack, 2023-05-09:
Here is a video of my experimentation with IR sensor.
https://www.youtube.com/watch?v=z8zI6PzrAY8

manuel-rw, 2023-06-06:
Thanks @bwack for answering.
I think I will stick with my hot air gun for now.
Closing this issue since it's resolved :)

closed 2023-06-06 by manuel-rw